ros2/rmw_connext #26: master does not compile on Windows

wjwwood, 2015-04-28:
See this latest job: http://54.183.26.131:8080/job/ros2_batch_ci_windows/199/console

This the relevant error:

```
"C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\build\userland\userland.sln" (default target) (1) ->
"C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\build\userland\add_two_ints_server.vcxproj.metaproj" (default target) (7) ->
"C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\build\userland\add_two_ints_server.vcxproj" (default target) (105) ->
(ClCompile target) -> 
 C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\src\ros2\examples\userland\src\add_two_ints_server.cpp(50): error C2668: 'rclcpp::node::Node::create_service': ambiguous call to overloaded function [C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\build\userland\add_two_ints_server.vcxproj]
```

It is repeated a few times. I guess this is something to do with the most recent changes. At first I thought it was the changes I was testing, but it turned out to be a problem on master too.

The commits tested are:

```
==> vcs log -l1 src
..................
[ ... ]
=== src\ros2\examples (git) ===
commit c7b5e7780170549dee4e36df394697d455f6ec03
Merge: 2472ad3 2a61abf
Author: Esteve Fernandez <esteve@apache.org>
Date:   Wed Apr 22 14:05:41 2015 -0700

    Merge pull request #15 from ros2/request-header
    
    Pass request header to callbacks
=== src\ros2\launch (git) ===
commit 47bb7510e0634acde3ac9048b41a525ae83cdc86
Author: Dirk Thomas <dthomas@osrfoundation.org>
Date:   Mon Apr 20 12:21:45 2015 -0700

    use waitpid when available
=== src\ros2\rcl (git) ===
commit a8962978a2f82059cef2097427e746e349a89d2a
Merge: 02db75d dd5f1de
Author: Dirk Thomas <dthomas@osrfoundation.org>
Date:   Tue Apr 7 17:04:52 2015 -0700

    Merge pull request #2 from ros2/code_style_uncrustify
    
    code style only
=== src\ros2\rcl_interfaces (git) ===
commit 645e40ebbfb8e857e46545b419965109989d4929
Author: Esteve Fernandez <esteve@osrfoundation.org>
Date:   Wed Apr 22 17:24:18 2015 -0700

    Renamed recurse to recursive
=== src\ros2\rclc (git) ===
commit a2b2292eb34f06bd931e1a70f84368e6c651cd3d
Merge: 9615893 9609603
Author: Dirk Thomas <dthomas@osrfoundation.org>
Date:   Tue Apr 7 17:04:57 2015 -0700

    Merge pull request #2 from ros2/code_style_uncrustify
    
    code style only
=== src\ros2\rclcpp (git) ===
commit 8ad1f1f4c5b1ab5b152cc664c56f3c991eaaac4f
Merge: 1bf595d 6b6b94f
Author: Esteve Fernandez <esteve@apache.org>
Date:   Tue Apr 28 15:09:22 2015 -0700

    Merge pull request #25 from ros2/spin-node-until-future-complete
    
    Added spin_node_until_future_complete
=== src\ros2\rmw (git) ===
commit 89900d73c6ed11c67b8b10e16a9eece4e2265629
Merge: 8524e33 889dce1
Author: Dirk Thomas <dthomas@osrfoundation.org>
Date:   Tue Apr 28 12:00:17 2015 -0700

    Merge pull request #11 from ros2/typesupport_for_rmw_impl
    
    export type support for rmw implementation
=== src\ros2\rmw_connext (git) ===
commit 1aece5657d69a34e8c808943105e3bb45a66c2ad
Author: Dirk Thomas <dthomas@osrfoundation.org>
Date:   Tue Apr 28 12:10:09 2015 -0700

    standardize target suffix
=== src\ros2\rmw_implementation (git) ===
commit d719244878056bc13b3c2381e97b1b618c4372d9
Author: Dirk Thomas <dthomas@osrfoundation.org>
Date:   Fri Apr 3 12:30:43 2015 -0700

    update license file to keep copyright template
=== src\ros2\rmw_opensplice (git) ===
commit 48e5f4f9dd32d49e2510b0ad99d5560f4fd2820a
Author: Dirk Thomas <dthomas@osrfoundation.org>
Date:   Tue Apr 28 12:10:02 2015 -0700

    standardize target suffix
=== src\ros2\rosidl (git) ===
commit 47cb8d2918c927956c556df959c4b46e89a3c57e
Author: Dirk Thomas <dthomas@osrfoundation.org>
Date:   Tue Apr 28 12:10:43 2015 -0700

    function to depend on include directories and libraries of generated interface target
=== src\ros2\rosidl_dds (git) ===
commit e749aecbcc49cc52d89ec0711b9a1af725a4b0eb
Author: Dirk Thomas <dthomas@osrfoundation.org>
Date:   Tue Apr 28 12:09:54 2015 -0700

    standardize target suffix
```

wjwwood, 2015-04-28:
@esteve, do you have any idea about this? A naive guess is that it has something to do with the ability to pass different kinds of callbacks to the `create_service` method.

esteve, 2015-04-28:
@wjwwood this should fix it, but it's too verbose IMO:

```cpp
  node->create_service<userland_msgs::AddTwoInts>(
    "service_name",
    static_cast<rclcpp::service::Service<userland_msgs::AddTwoInts>::CallbackWithHeaderType>(add));
```

wjwwood, 2015-04-29:
I'm not sure why I created this issue here, probably should be on `rclcpp`.

Anyways, your suggestion works, but I agree that it's too verbose:

```diff
diff --git a/userland/src/add_two_ints_server.cpp b/userland/src/add_two_ints_server.cpp
index 72a3573..377f5b8 100644
--- a/userland/src/add_two_ints_server.cpp
+++ b/userland/src/add_two_ints_server.cpp
@@ -47,7 +47,8 @@ int main(int argc, char ** argv)

   auto node = rclcpp::Node::make_shared("add_two_ints_server");

-  node->create_service<userland_msgs::AddTwoInts>("add_two_ints", add);
+  node->create_service<userland_msgs::AddTwoInts>("add_two_ints",
+    static_cast<rclcpp::service::Service<userland_msgs::AddTwoInts>::CallbackWithHeaderType>(add));

   rclcpp::spin(node);

diff --git a/userland/test/test_server.cpp b/userland/test/test_server.cpp
index 612e449..8190fdf 100644
--- a/userland/test/test_server.cpp
+++ b/userland/test/test_server.cpp
@@ -33,7 +33,8 @@ int main(int argc, char ** argv)

   auto node = rclcpp::Node::make_shared("test_server");

-  node->create_service<userland_msgs::AddTwoInts>("service_name", add);
+  node->create_service<userland_msgs::AddTwoInts>("service_name",
+    static_cast<rclcpp::service::Service<userland_msgs::AddTwoInts>::CallbackWithHeaderType>(add));

   rclcpp::spin(node);

```

I'm going to try and resolve it some other way.

closed 2015-04-29 by wjwwood