ros2/rmw_connext #63: fix cleanup functions

dirk-thomas, 2015-07-29:
In #45 various clean up functions have been introduced. In various tests (e.g. in `test_rclcpp`) these functions print error messages:

    Error in destruction of rmw publisher handle: failed to delete contained entities for publisher, at ...
    Error in destruction of rmw subscription handle: failed to delete subscriber, at ...

wjwwood, 2015-07-30:
I guess I didn't notice the output of the tests since they were passing. I would expect that to cause a failure (return code != 0 or exception) since those messages occur the `rmw_*_destroy` functions. Probably they're being checked in a C++ destructor and being printed to the screen rather than throwing to avoid throwing in a destructor.

jacquelinekay, 2015-11-11:
I'm running into an issue where a test case that initializes services/clients in a gtest with prior cases with pub/sub does not send any messages. This may be related, so I'm going to investigate cleanup issues in rmw and see if it helps.

dirk-thomas, 2016-02-13:
Furthermore the Connext typesupport  does currently not provide any destroy functions for servers / clients.

closed 2023-06-21 by clalancette