ros2/rmw_connext #76: Pub/sub fails across different nodes in same process

jacquelinekay, 2015-08-13:
Branch multiple_nodes in system_tests, package test_rclcpp, illustrates this bug:
https://github.com/ros2/system_tests/tree/multiple_nodes

Case 1 fails for Connext and passes for Opensplice:
node1 and node2 are both added to one executor.
node1 publishes "foo", node2 subscribes to "foo"
node2 publishes "bar", node2 subscribes to "bar"
Both publishers publish 5 times.
0/5 messages are received for both subscribers.

Case 2 passes:
one node publishes "foo", "bar", subscribes to "foo" and "bar"
Both publishers publish 5 times.
5/5 messages are received for both subscribers.

Case 3 passes:
node1 and node2 both added to one executor.
node1 publishes "foo", subscribes to "foo"
node2 publishes "bar", subscribes to "bar"
Both publishers publish 5 times.
5/5 messages are received for both subscribers.

wjwwood, 2015-08-13:
Is it specific to connext or does it include connext dynamic?

jacquelinekay, 2015-08-13:
Both connext and connext_dynamic fail.

wjwwood, 2015-08-13:
Cool, I actually want this behavior for intra process so I'm supper interested in how to duplicate it :)

jacquelinekay, 2015-08-14:
Output from `rtiddsspy` for the `spin_across_nodes` test case:

```
1439516100.176273  W +N  C0A86401    foo                 test_rclcpp::msg::
                                ...                      dds_::UInt32_     
1439516100.177054  R +N  C0A86401    bar                 test_rclcpp::msg::
                                ...                      dds_::UInt32_     
1439516100.177273  d +N  C0A86401    foo                 test_rclcpp::msg::
                                ...                      dds_::UInt32_     
1439516100.177411  d +M  C0A86401    foo                 test_rclcpp::msg::
                                ...                      dds_::UInt32_     
1439516100.177484  d +M  C0A86401    foo                 test_rclcpp::msg::
                                ...                      dds_::UInt32_     
1439516100.177553  d +M  C0A86401    foo                 test_rclcpp::msg::
                                ...                      dds_::UInt32_     
1439516100.177623  d +M  C0A86401    foo                 test_rclcpp::msg::
                                ...                      dds_::UInt32_     
1439516100.177893  R -M  C0A86401    bar                 test_rclcpp::msg::
                                ...                      dds_::UInt32_     
(time invalid)     d ?M  C0A86401    foo                 test_rclcpp::msg::
                                ...                      dds_::UInt32_     
1439516100.178160  W -M  C0A86401    foo                 test_rclcpp::msg::
                                ...                      dds_::UInt32_    
```

So, the "foo" writer gets discovered, the "bar" reader gets discovered, and the five "foo" messages make it to the DDS layer.

But, the "foo" reader doesn't get discovered and the "bar" writer doesn't get discovered, hence there are no "bar" messages published over DDS and rcl doesn't receive the "foo" messages because there's no DDS reader.

jacquelinekay, 2015-08-14:
If I switch the order of instantiation of the nodes, the topics also switched (i.e. the "bar" reader is missing and the "foo" writer is missing)

jacquelinekay, 2015-10-13:
I will investigate this further and clarify if it's actually an issue, and close the issue if it's not,

jacquelinekay, 2015-10-13:
going to close and reopen this as a new issue.

closed 2015-10-13 by jacquelinekay