ros2/rmw_connext #78: Import of symbols is not working fine on 5.2 Community and VS2015

j-rivero, 2015-08-14:
We need to use some flags to get proper visibility from headers in the 5.2 Community version of RTI connext. The following patch implement the ones needed in `rmw_connext` and `rmw_connext_dynamic`.

diff --git a/connext_cmake_module/cmake/Modules/FindConnext.cmake b/connext_cmak
index 087a563..aac2436 100644
--- a/connext_cmake_module/cmake/Modules/FindConnext.cmake
+++ b/connext_cmake_module/cmake/Modules/FindConnext.cmake
@@ -183,7 +183,12 @@ if(NOT "${_NDDSHOME} " STREQUAL " ")
   set(Connext_LIBRARY_DIR "${Connext_LIBRARY_DIRS}")

   if(WIN32)
-    set(Connext_DEFINITIONS "RTI_WIN32" "NDDS_DLL_VARIABLE")
+    set(Connext_DEFINITIONS "RTI_WIN32"
+                            "NDDS_DLL_VARIABLE"
+                            "RTI_dds_c_DLL_VARIABLE"
+                            "RTI_dds_cpp_DLL_VARIABLE"
+                            "RTI_log_DLL_VARIABLE")
+
     # This will be a .bat file and it will be on the PATH.
     set(Connext_DDSGEN2 "rtiddsgen.bat")
   else()

dirk-thomas, 2015-08-14:
For the professional version this is not necessary since the `ndds_cpp.h` file does the following:

    #if defined(_WIN32) && !defined(RTI_WINCE) && _MSC_VER > 1310 && !defined(RTI_WIN32)
      #define RTI_WIN32
      #ifdef _DLL /* compiling against DLL version of the CRT */
        #define NDDS_DLL_VARIABLE
        ...

In the community version I didn't find how `RTI_WIN32` and `NDDS_DLL_VARIABLE` is set (I assume that's why we set them in the `Connext_DEFINITIONS`. All other variables are derived from that and should not need to be set manually, see https://github.com/rticommunity/rticonnextdds-src/blob/master/dds_cpp.2.0/hpp/dds/core/detail/macros.hpp#L28-L46

j-rivero, 2015-08-17:
Thanks for the comparison Dirk. I've set the patch to upstream in order to properly defining the flags like it is done in Professional version. Once the PR is merged, we can remove the definitions in the FindConnext.cmake module.

dirk-thomas, 2015-10-13:
@j-rivero What is the status of this PR? Is it obsolete with the PR upstream being merged?

j-rivero, 2015-10-13:
> @j-rivero What is the status of this PR? Is it obsolete with the PR upstream being merged?

Right. Sorry, closing.

closed 2015-10-13 by j-rivero