ros2/rmw_connext #80: Update the hook to call 5.2 instead 5.1

j-rivero, 2015-08-14:
Simple fix to call the 5.2 instead of 5.1

```patch
--- a/connext_cmake_module/env_hook/connext.bat.in
+++ b/connext_cmake_module/env_hook/connext.bat.in
@@ -1,7 +1,7 @@
 set "Connext_HOME=@Connext_HOME@"
 
 :: Call RTI's env setup script, piping stdout to nul, since they have echo on.
-call "%Connext_HOME:/=\%\..\rti_set_env_5.1.0.bat" 1> nul
+call "%Connext_HOME:/=\%\..\rti_set_env_5.2.0.bat" 1> nul
 
 :: Add the Connext_LIBRARY_DIR to the Path so .dll's can be found at runtime.
 set "Connext_LIBRARY_DIR=@Connext_LIBRARY_DIR@"
```

wjwwood, 2015-08-14:
@j-rivero can you please open pull requests for this? You can even edit files directly and then propose a pull request based on that edit, for example: https://github.com/ros2/rmw_connext/edit/master/connext_cmake_module/env_hook/connext.bat.in

This will make it easier for us to just merge a fix like this without loosing attribution information.

dirk-thomas, 2015-08-14:
In Connext 5.2 Pro the file is now called `rtisetenv_X64Win64VS2013.bat` (see ros2/ros2#84). We should aim to make our custom build look similar.

j-rivero, 2015-08-19:
Thanks guys, pull request #87

dirk-thomas, 2015-10-22:
Addressed by #104.

closed 2015-10-22 by dirk-thomas