ros2/rmw_connext #86: Setup windows debug libraries in Connext_LIBRARIES (cmake module)

j-rivero, 2015-08-19:
The Visual Studio build of RTI Connext generates release and debug libraries (same using a "d" postfix). The issues is not critical as far as you don't need to debug the RTI Connext libraries. And the workaround would be to modify the `_excpected_library_base_names` in `FindConnext.cmake` package adding a `d` at the end.

I would say that the proper way of getting this defined in the Find cmake module would be to use:

```cmake
set(Connext_LIBRARIES optimized nddsc debug nddscd
                      optimized ...   debug ...)
```
Something similar is implemented in the [FindwxWindows.cmake module](https://github.com/Kitware/CMake/blob/master/Modules/FindwxWindows.cmake#L403)

When I tried a quick test for this layout ament is generating the error:

```
ament_export_libraries() package 'rmw_connext_cpp' passes the build configuration keyword 'debug' as the last exported library
```

Could be that some parsing/code in the ament package does not support this way of defining the libraries? I need to investigate about it.

dirk-thomas, 2015-08-19:
ament supports build configuration keywords. If you can point me to the modified version and what exactly you called I could take a look.

j-rivero, 2015-08-19:
> If you can point me to the modified version and what exactly you called I could take a look.

I run a quick test just adding this line at the end of the FindConnext.cmake module (just a stub for triggering the error).

```patch
@@ -249,6 +228,8 @@ if(Connext_FOUND AND NOT WIN32)
   list(APPEND Connext_LIBRARIES "pthread" "dl")
 endif()

+set(Connext_LIBRARIES optimized nddsc debug nddscd)
+
 include(FindPackageHandleStandardArgs)
 # Connext_HOME, Connext_LIBRARY_DIRS, and Connext_LIBRARY_DIR are not always se
 find_package_handle_standard_args(Connext
```

Install the connext_cmake_module and run `ament build --only rmw_connext_cpp`

```
 - Found rosidl_typesupport_connext_cpp: 0.0.0 (C:/Users/jrivero/code/ros2_build
/install/share/rosidl_typesupport_connext_cpp/cmake)
CMake Error at C:/Users/jrivero/code/ros2_build/install/share/ament_cmake_export
_libraries/cmake/ament_export_libraries.cmake:48 (message):
  ament_export_libraries() package 'rmw_connext_cpp' passes the build
  configuration keyword 'optimized' as the last exported library
Call Stack (most recent call first):
  CMakeLists.txt:44 (ament_export_libraries)

-- Configuring incomplete, errors occurred!
See also "C:/Users/jrivero/code/ros2_build/build/rmw_connext_cpp/CMakeFiles/CMak
eOutput.log".
```

dirk-thomas, 2015-08-20:
That is a quoting problem in the `rmw_connext_cpp` CMake code. The following PR fixes this: ros2/rmw_connext#90

PR #123 (2016-02-09): export optimized as well as debug libraries if found
Fixes #86.

Before: http://ci.ros2.org/job/ci_windows/994/
After: http://ci.ros2.org/job/ci_windows/997/
closed 2016-02-09 by dirk-thomas