ros2/rmw_connext #125: Fixed guard conditions are not reset

gerkey, 2016-02-09:
@dirk-thomas and I identified an apparent bug: while the non-fixed guard conditions are [reset to false after waiting](https://github.com/ros2/rmw_connext/blob/master/rmw_connext_shared_cpp/include/rmw_connext_shared_cpp/shared_functions.hpp#L406-L434), the fixed guard conditions that were [added during waitset creation](https://github.com/ros2/rmw_connext/blob/master/rmw_connext_shared_cpp/src/shared_functions.cpp#L582-L601) are never reset to false.

As a result, if you've ever done something to trigger a fixed guard condition (e.g., add a node or subscriber), then `rmw_wait()` on Connext should return immediately, every time you call it, without waiting. That behavior might explain many of our test failures.

The proposed fix is to add a block to the end of the wait call to reset the fixed guard conditions.

jacquelinekay, 2016-02-09:
This is probably my fault. Let me know if I can help.

dirk-thomas, 2016-02-09:
The same bug might also be in the other rmw implementations.

PR #129 (2016-02-11): Reset fixed guard conditions
Should fix #125.

CI jobs (connext-only) are in the queue.
closed 2016-02-11 by gerkey