ros2/rmw_connext #158: Which version of Connext DDS is used/supported in ROS 2

firesurfer, 2016-05-05:
Due to problems with Opensplice I'm evaluating other DDS implementations.

I took a closer look at https://www.rti.com/products/ .
There are several versions of Connext DDS like Professional, Secure, Micro, Cert. 
Which of them is supported with ros 2 ?

As far as i can see rti connext is available for raspberry pis. ( https://community.rti.com/content/forum-topic/howto-run-rti-connext-dds-raspberry-pi )
Do you know of someone who already tried to run ROS 2 with rit connext on a raspberry pi?

dirk-thomas, 2016-05-05:
We use the professional version for local testing and CI.

We haven't heard from anyone using connect on a raspberry pi.

dirk-thomas, 2016-05-18:
I guess this has been answered so I will close the ticket.

closed 2016-05-18 by dirk-thomas