ros2/rmw_connext #197: Appropriate placement of generated idls for rmw_connext

jonbonne, 2016-08-26:
Currently the request/response rtiddsgen generated idls are being installed in the "<pkg_name>/msg/dds_connext" folder. They should be appropriately placed in the "<pkg_name>/srv/dds_connext" folder.

wjwwood, 2016-08-26:
Perhaps they should. It might be the way it is right now, just as a matter of convenience since services are just generated into two messages (the request and response messages). Either way, it should be both possible and more correct to do as you've suggested.

I'm not sure when one of us will have time to address this. The code surrounding that is fairly complicated, but if you're feeling bold we'd be happy to help you get a pull request merged that does this. 😄

dirk-thomas, 2016-09-14:
@jonbonne Thank you for reporting the issue.

While the generated request and response interfaces already use the `srv` namespace and are also being generated in the `srv` subfolder in the build space the installation rules then put them into the `msg` subfolder which is indeed wrong. The `.idl` are already in the right location as far as I can see.

Please see the referenced PRs which address the problem.

* Linux [image]
* Linux [image]

closed 2016-09-19 by dirk-thomas