ros2/rmw_connext #204: [connext_dynamic] currently broken

dirk-thomas, 2016-11-02:
As a regression of #194 the dynamic rmw implementation doesn't function anymore. Even for simple pub/sub examples the executables crash in the new `add_information` calls.

Karsten1987, 2017-04-11:
As far as I can tell, the current master branch does not even compile. Is it worth putting an AMENT_IGNORE in it until it's fixed?

mikaelarguedas, 2017-04-11:
Slightly more context: broken since the type change (char => signed char) introduced by https://github.com/ros2/rosidl/pull/211.
We should decide if we should drop AMENT_IGNORE files in all not currently supported  rmw_implementations (and remove them in ci when needed) or any other solution to avoid users checking out the ros2.repos file to have a workspace not compiling

dirk-thomas, 2017-04-11:
We have talked about this case in the last meeting. I still think we should keep the unsupported rmw impl. as close in sync as possible. If we don't the technical debt while pile up and it will "cost" us much more to revisit this in the future. Especially when the patch is as simple as #222 (and it is easy now since all our memory about recent changes is fresh).

mikaelarguedas, 2017-04-11:
closed via https://github.com/ros2/rmw_connext/pull/222

dirk-thomas, 2017-04-11:
This is not fixed. `rmw_connext_dynamic_cpp` still doesn't "work" - it only compiles again. Reopened.

mikaelarguedas, 2017-04-11:
oh sorry I thought the scope of this was "compilation broken" not "runtime broken", thanks for catching that

closed 2023-06-21 by clalancette