ros2/rmw_connext #205: [connext_dynamic_cpp] Can't send/receive StaticArrayNested using Python / C typesupport

mikaelarguedas, 2016-11-02:
This has always been the case and is not related to the recent breakage.
Test blacklisted [here](https://github.com/ros2/system_tests/pull/152/commits/491d716941994590efc0e9a1fdfe72d0445fba5c#diff-02dc43db6f966560e7ee414c298df21eR144)

dirk-thomas, 2016-11-02:
Since the dynamic implementation is not being tested by default I would rather not disable those tests. Instead when it is enabled it would be helpful to see what is actually not working / implemented.

mikaelarguedas, 2016-11-02:
Yeah I disabled it when it was still being tested by our CI.
Reenabled the failing test in https://github.com/ros2/system_tests/commit/0dc901f78ee1cb078966c6a4010bc33c35016cd6

closed 2023-06-21 by clalancette