ros2/rmw_connext #223: RTI shared memory issue

Karsten1987, 2017-04-11:
I am currently testing the namespace implementation on connext.
[image]

 On the build farm i get rti related error messages, which I cannot reproduce locally:
```
18:10:14 4: [test_executable_0] [D0108|ENABLE]RTIOsapiSharedMemoryMutex_create:OS semget() failure, error 0X1C: No space left on device
18:10:14 4: [test_executable_0] [D0108|ENABLE]NDDS_Transport_Shmem_create_recvresource_rrEA:failed to initialize shared memory resource mutex for key 0xb086aa
```

The RTI knowledge base recommends increasing the number of allowed semaphores and such, however I am a bit sceptical about it. Has one of you ever encountered similar behavior?

https://research.rti.com/kb/what-causes-rtiosapisharedmemorymutexcreateos-semget-failure-error-0x1c-error-message

Karsten1987, 2017-04-11:
reading further here:
http://community.rti.com/rti-doc/45c/ndds.4.5c/doc/html/api_c/group__NDDS__Transport__Shmem__Plugin.html

> Crashing and Restarting Programs
> 
> If a process using shared memory crashes (say because the user typed in ^C), resources associated with its shared memory ports may not be properly cleaned up. Later, if another RTI Data Distribution Service process needs to open the same ports (say, the crashed program is restarted), it will attempt to reuse the shared memory segment left behind by the crashed process.
> The reuse is allowed iff the properties of transport plugin are compatible with those embedded in the shared memory segment (i.e., of the original creator). Otherwise, the process will fail to open the ports, and will output messages such as below (with numbers appropriate to the properties of the transport plugins involved).
> 
>   NDDS_Transport_Shmem_create_recvresource_rrEA:failed to initialize shared 
>   	memory resource Cannot recycle existing shmem: size not compatible for key 0x1234 		  
> In this scenario, the shared memory segments must be cleaned up using appropriate platform specific commands. For details, please refer to the platform notes.

dirk-thomas, 2017-04-11:
See below a snippet from a previous conversation on our internal mailing list. I am not sure if that has been applied to more machines and/or documented anywhere.

---

On Thu, Dec 31, 2015 at 1:18 PM, Brian Gerkey <gerkey@osrfoundation.org> wrote:
This test failure:

http://ci.ros2.org/view/nightly/job/nightly_osx/163/testReport/projectroot.Users.osrf.jenkins.workspace.nightly_osx.ws.src.ros2.examples/rclcpp_examples/test_example_talker_listener__rmw_connext_cpp/

Includes this unexpected output:

~~~
[test_executable_0]
[D0111|ENABLE]NDDS_Transport_Shmem_create_recvresource_rrEA:failed to
initialize shared memory resource segment for key 0x408972
[test_executable_0]
[D0111|ENABLE]NDDS_Transport_Shmem_create_recvresource_rrEA:It is
possible that your system's shared memory settings are not sufficient
to run RTI Connext DDS. For a more in-depth explanation of the problem
and solution, please visit http://community.rti.com/kb/osx510
[test_executable_0]
[D0111|ENABLE]NDDS_Transport_Shmem_create_recvresource_rrEA:!give
mutex
[test_executable_0]
[D0111|ENABLE]NDDS_Transport_Shmem_create_recvresource_rrEA:!give
mutex
[test_executable_0]
[D0111|ENABLE]NDDS_Transport_Shmem_create_recvresource_rrEA:failed to
initialize recvResource for port 0x8974
~~~

On dosa, I can see in /etc/sysctl.conf that we've already done what
RTI recommended to address this issue
(https://community.rti.com/kb/osx510). On the assumption that we're
sometimes in need of more shared memory than is allowed by their
recommendation, I heuristically increased kern.sysv.shmmax on dosa
from 419MB to 1G (1073741824) and rebooted the machine.

mikaelarguedas, 2017-05-10:
@Karsten1987 to update this

Karsten1987, 2017-06-07:
I'll close this for now as I don't run into this situation anymore.

However, when I was debugging, the configuration which seemed to help was:
```
docker run --privileged --shm-size=1G --sysctl kernel.shmmax=1073741824 --sysctl kernel.shmmni=1024 --sysctl kernel.sem="250 32000 100 1024" -e UID=`id -u` -e GID=`id -g` -e CI_ARGS="$CI_ARGS" -e CCACHE_DIR=/home/rosbuild/.ccache -i -v `pwd`:/home/rosbuild/ci_scripts -v $HOME/.ccache:/home/rosbuild/.ccache $CONTAINER_NAME
```

Please find the complete configuration here: http://ci.ros2.org/view/All/job/test_ci_linux/configure

closed 2017-06-07 by Karsten1987