ros2/rmw_connext #233: Testing for Beta2 coverage demo_nodes_cpp parameter tools are not working correctly

tfoote, 2017-06-29:
I am running the parameter nodes in demo_nodes_cpp and under connext they are not working correctly. They appear to be hanging. The events executables provide some output after the Ctrl-C but not the full equivalent of the fastrtps runs. I've waited 30+ seconds and it still responds quickly immediately after the Ctrl-C.

Output of events after Ctrl-C:
[image]
Async version: 
[image]
Simple set and get just hangs:
[image]
For reference fastrtps running the same sample in the same workspace:
[image]

tfoote, 2017-06-29:
I can get the following errors if I run parameter_list side by side simultaneously. This might be the different impelementations. Did we ever homogenize the parameter implementation?

```
C:\dev\ros2>ros2 run demo_nodes_cpp list_parameters
RTI Data Distribution Service Evaluation License issued to OSRF - OSRF01 dthomas@osrfoundation.org For non-production use only.
Expires on 05-may-2018 See www.rti.com for more information.
PRESParticipant_getLocalOrRemoteParticipantShmemTransportCompatibility:!goto key remoteParticipant
PRESPsService_linkToLocalWriter:!getLocalOrRemoteParticipantShmemTransportCompatibility
PRESPsService_onLinkToLocalEndpointEvent:!create linkToLocalWriter
PRESParticipant_getLocalOrRemoteParticipantShmemTransportCompatibility:!goto key remoteParticipant
PRESPsService_linkToLocalWriter:!getLocalOrRemoteParticipantShmemTransportCompatibility
PRESPsService_onLinkToLocalEndpointEvent:!create linkToLocalWriter
PRESParticipant_getLocalOrRemoteParticipantShmemTransportCompatibility:!goto key remoteParticipant
PRESPsService_linkToLocalWriter:!getLocalOrRemoteParticipantShmemTransportCompatibility
PRESPsService_onLinkToLocalEndpointEvent:!create linkToLocalWriter
signal_handler(2)
```

[image]

tfoote, 2017-06-29:
ros2param has a different error:
```
C:\dev\ros2>ros2 run demo_nodes_cpp ros2param list add_two_ints_server
RTI Data Distribution Service Evaluation License issued to OSRF - OSRF01 dthomas@osrfoundation.org For non-production use only.
Expires on 05-may-2018 See www.rti.com for more information.
Timed out trying to list parameters: 10 seconds

C:\dev\ros2>
```

dirk-thomas, 2017-06-29:
I tried to run all executables you mentioned (`parameter_events`, `parameter_events_async`, `set_and_get_parameters`, `list_parameters`) with Connext and all of them worked fine for me. I used a from-source build from on of the Jenkins nodes.

Please try to reproduce the problem from a fresh build and provide more information.

tfoote, 2017-06-29:
Closing this as I can't reproduce today with an updated tree.

closed 2017-06-29 by tfoote