ros2/rmw_connext #246: catch error before nullptr access segfault

serge-nikulin, 2017-09-11:
## Bug report

**Required Info:**

- Operating System: Windows 10
  - 
- Installation type: from source
  - 
- Version or commit hash: 188147a0428d199a1924764fa00df177549b6149
  - 
- DDS implementation: RTI Connext
  - 
- Client library (if applicable): N/A
  - 

#### Steps to reproduce issue
 Visual code inspection
```

```

#### Expected behavior
Return value of _DataReader::narrow call should be checked for nullptr
#### Actual behavior
rosidl_typesupport_connext_cpp/resource/msg__type_support.cpp.em(167): no check
rosidl_typesupport_connext_cpp/resource/msg__type_support.cpp.em(251): no check
rosidl_typesupport_connext_c/resource/msg__type_support_c.cpp.em(279): [OK, see this commit ](https://github.com/ros2/rmw_connext/commit/9ff32755d16088c432c541477bc158dfe5a2a255)
rosidl_typesupport_connext_c/resource/msg__type_support_c.cpp.em(439): no check

#### Additional information
In my previous company I have seen RTI Connext fails on this call. I highly recommend to test all returned pointers for NULL.

----
## Feature request

#### Feature description

#### Implementation considerations
Please repeat this fix for all narrow calls:
https://github.com/ros2/rmw_connext/commit/9ff32755d16088c432c541477bc158dfe5a2a255

dirk-thomas, 2017-09-11:
These cases should indeed be catched. Thanks for pointing it out.

Please consider to provide a pull request to improve this.

serge-nikulin, 2017-09-11:
Why enhancement and not a bug?

serge-nikulin, 2017-09-11:
@dejanpan, FYI

dirk-thomas, 2017-09-11:
I labeled it "enhancement" since it is currently not resulting in any problems as far as we know. You can argue that it should have been written "defensive" in the first place (which I agree with).

serge-nikulin, 2017-09-11:
Sorry about that, in my previous company all unchecked pointers were treated as compilation errors.

serge-nikulin, 2017-09-12:
Pull request https://github.com/ros2/rmw_connext/pull/247

dirk-thomas, 2017-09-14:
I will close this since it was addressed by #247.

@serge-nikulin If you include the following text in your PR it will automatically close this issue when the PR gets merged:

> Fixes #246

closed 2017-09-14 by dirk-thomas