ros2/rmw_connext #248: Wrong WIN32 compiler definiton

serge-nikulin, 2017-09-13:
## Bug report

**Required Info:**

- Operating System: Windows 10
  - 
- Installation type: from source
  - 
- Version or commit hash: 188147a0428d199a1924764fa00df177549b6149
  - 
- DDS implementation: RTI Connext
  - 
- Client library (if applicable): N/A
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior
rmw_connext_cpp/CMakeLists.txt, lines 116 and 122 should contain RMW_CONNEXT_CPP_BUILDING_DLL definition.

#### Actual behavior
The lines use ROSIDL_TYPESUPPORT_CONNEXT_CPP_BUILDING_DLL definition (copy paste error?)

serge-nikulin, 2017-09-13:
@dejanpan, FYI

dirk-thomas, 2017-09-13:
Thanks for pointing this out. The usage of `ROSIDL_TYPESUPPORT_CONNEXT_CPP_BUILDING_DLL` in the `rmw_connext_cpp` CMake file is a left over of previous iterations of the typesupport and is unnecessary. Removed in #249.

`RMW_CONNEXT_CPP_BUILDING_DLL` is already being defined in https://github.com/ros2/rmw_connext/blob/188147a0428d199a1924764fa00df177549b6149/rmw_connext_cpp/CMakeLists.txt#L110-L111 otherwise this wouldn't work.

mikaelarguedas, 2017-09-14:
@dirk-thomas Does this mean that the same fix can be applied to opensplice ?

dirk-thomas, 2017-09-14:
> Does this mean that the same fix can be applied to opensplice ?

Maybe, it depends on how those libraries / headers are being used by each other. You could give it a try - each rmw impl. and typesupport is slightly different.

PR #249 (2017-09-14): remove unnecessary define
Fixes #248.

* Linux [image]
* Windows [image]

Waiting with the merge until after the Beta 3 release.
closed 2017-09-14 by dirk-thomas