ros2/rmw_connext #254: FindConnext much slower since 5.3.0 upgrade

mikaelarguedas, 2017-10-03:
Finding Connext was taking a few seconds and is now taking ~30sec since we upgraded to Connext 5.3.0.
The main reason seems to be the execution time of `rtiddsgen_server` invoked from CMake.

See https://github.com/ros2/rmw_connext/pull/253#issuecomment-333705170 for more details

dirk-thomas, 2017-10-04:
I have filled a ticket with RTI since the problem seems to be that when being invoked from CMake the script waits until the forked daemon times out (which defaults to 20s). So the forking logic must have changed in Connext 5.3 which is responsible for the regression.

mikaelarguedas, 2017-10-05:
To be exact the behavior did not change between 5.2 and 5.3. The only thing is that, in 5.2, when called without any argument it was failing early and not forking the daemon. In 5.3 it forks the daemon regardless of the presence of arguments. Since #253 I would expect the same behavior (waiting for the daemon to time out) for 5.2 and 5.3

PR #255 (2017-10-06): fix hang of server waiting for forked daemon to time out
Temporarily fixes #254.
closed 2017-10-06 by dirk-thomas