ros2/rmw_connext #263: Code in wait function continues to possibly return RMW_RET_OK despite DDS return error

dejanpan, 2017-10-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04

- Installation type:
   -
 (https://github.com/ros2/rmw_connext/blob/661993abf92996213576fccf6621aacd54fa5023/rmw_connext_shared_cpp/include/rmw_connext_shared_cpp/wait.hpp#L254)
  -
 (https://github.com/ros2/rmw_connext/blob/661993abf92996213576fccf6621aacd54fa5023/rmw_connext_shared_cpp/include/rmw_connext_shared_cpp/wait.hpp#L288)
  -
 (https://github.com/ros2/rmw_connext/blob/661993abf92996213576fccf6621aacd54fa5023/rmw_connext_shared_cpp/include/rmw_connext_shared_cpp/wait.hpp#L321)
  -
 (https://github.com/ros2/rmw_connext/blob/661993abf92996213576fccf6621aacd54fa5023/rmw_connext_shared_cpp/include/rmw_connext_shared_cpp/wait.hpp#L353)
- Version or commit hash:
  - see above
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - NA

#### Steps to reproduce issue
Quoting @dirk-thomas 
```
After setting the error message the function should return `RMW_RET_ERROR`. The quick fix would be to just add the return statement in the cases it is missing.

The "correct" fix would be to update the structure of the code to store the return value in a value but still perform the other cleanup and only at the very end `return`.
```

I will provide PR for the quick fix and started an issue for the `correct` fix.

dejanpan, 2017-10-28:
@dirk-thomas here we are also missing a `return;` right?

dirk-thomas, 2017-10-30:
> [here](https://github.com/ros2/rmw_opensplice/blob/master/rmw_opensplice_cpp/src/rmw_wait.cpp#L99) we are also missing a `return;` right?

Yes, indeed. Since I saw and merged the PR first can you please create a follow up PR for that line. Thanks.

closed 2017-10-30 by dirk-thomas