ros2/rmw_connext #324: ros2 topic * commands can't always determine the type with Connext

clalancette, 2018-12-14:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04 AMD64
- Installation type:
  - Source
- Version or commit hash:
  - Release versions of crystal, from https://github.com/ros2/ros2/pull/624
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - rclpy-ish

#### Steps to reproduce issue
Terminal 1:
```
RMW_IMPLEMENTATION=rmw_connext_cpp ros2 run demo_nodes_cpp talker
```

Terminal 2:
```
RMW_IMPLEMENTATION=rmw_connext_cpp ros2 topic echo /chatter
```

#### Expected behavior
The data on the /chatter topic is printed.

#### Actual behavior
Around 20% of the time, the `ros2 topic` command will hang for 3-5 seconds, then print:

```
Could not determine the type for the passed topic
```

#### Additional information
Re-running the relevant command usually (but not always) makes it start working.  Running it a number of times in a row always makes it start running.

Also, I may have seen this with other RMW implementations (particularly Fast-RTPS), but it seems much more common on Connext (though I haven't yet ruled out other factors).

AwooOOoo, 2019-06-26:
I have seen the same with opensplice DDS.

dawonn-haval, 2020-05-09:
I have this issues on fast_rtps, opensplice, connext, and cyclone. 

Sometimes it works, sometimes not.  I don't know where to start debugging this.
rostopic list shows the topic, and if I specify the type it will sometimes echo ok. Other times nothing.

clalancette, 2020-05-11:
Generally, the problem revolves around the fact that `ros2 topic echo` creates a [new node](https://github.com/ros2/ros2cli/blob/cab0dcfbf0f90991fc7f810aa5989b92c12027a6/ros2topic/ros2topic/verb/echo.py#L95) every time it is run.  The consequence of that is that it needs to "join" the DDS graph, and doing so can take some time.  If that takes too long, then when it goes to get the available topic names and types, it can't find the one requested and prints that error.

We've talked a bit about changing this to use the `ros2 daemon` as well.  That is, when it first started, `ros2 topic echo` would attempt to contact the daemon and ask it for its list of topics and types.  If it gets a response, then it would go looking for the correct one.  If it doesn't get a response, then it can fall back to the old method of creating a node and joining the graph itself.

We just haven't had the time to implement this idea.  If anyone has time and would like to open a PR for this, we would happily review it.

clalancette, 2020-05-11:
Ha, I forgot that I had opened a more generic issue about this: https://github.com/ros2/ros2cli/issues/488 .  I'm going to close this one in favor of that one.

closed 2020-05-11 by clalancette