ros2/rmw_connext #326: Revisit DDS_Guid_t based caching

jwillemsen, 2018-12-17:
## Feature request

#### Feature description
The code currently uses DDS_Guid_t for caching which is constructed with some special code from DDS_InstanceHandle_t. We would expect user based caching to be based on DDS_InstanceHandle_t which is given back from the DDS API. The DDS_Guid_t is a very low level type and normally isn't used by users of DDS. Also the conversion from DDS_InstanceHandle_t to DDS_Guid_t doesn't work with all vendors, for example OpenDDS has a long for DDS_InstanceHandle_t. 

#### Implementation considerations
Maybe caching can be done based on DDS_InstanceHandle_t which is the type normally seen by users

jwillemsen, 2019-02-22:
Want to note that with OpenDDS the DDS::InstanceHandle_t is only unique within one DDS DomainParticipant.

wjwwood, 2019-02-23:
@jwillemsen even if the instance handle is only unique within a participant, they can be used to represent remote entities right? For instance when you get the same info for a sample, the publication_handle (of type `InstanceHandle_t`) with in the same info represents the remote data writer and does not collide with data writers which are owned by the local participant. That's the only purpose we use the instance handles for, keep track of remote entities, we never compare them (as far as I'm aware) between participants.

I believe the guid is used as a key (I assume that's what you're referring to) in a map because I think it wasn't easy to use the instance handle because it didn't have a convenient comparator to use. If that's not the case you're referring to, please link to specific lines in the code that are an issue.

dirk-thomas, 2019-11-11:
Closing due to no further activity.

closed 2019-11-11 by dirk-thomas