ros2/rmw_connext #327: Refactor rwm_connect into rmw_connect and rmw_dds

jwillemsen, 2018-12-17:
## Feature request

#### Feature description
For IDL to C++ based DDS implementations there is no need for making a complete rmw_ library to integrate one vendor. By using a generic rmw_dds library vendors like connext, splice, opendds can be integrated with much less code duplication. We have done this for connext and opendds as part of our [AXCIOMA](http://www.axcioma.com) product, it took us several iterations but at the end the core differences are minimal between DDS vendors

Using some templates or helper classes the minimal changes between vendors can be handled, these changes are in the area of:
- Retrieval of the domain participant factory
- QoS XML usage

#### Implementation considerations

adamsj-oci, 2019-11-22:
@jwillemsen, did you mean to call it rm_connect in the title? I'm assuming you meant rmw_connext. It probably won't make a difference but might be helpful for searching, etc to change it.

dirk-thomas, 2019-11-22:
While this would be a good goal for a refactoring there are difference in the API of each vendor which will likely be a problem.

Also since nobody is planning to work on this and it had been about a year since this ticket was filled I will go ahead and close the ticket. Please feel free to continue commenting. Any pull request towards this goal would be appreciated.

closed 2019-11-22 by dirk-thomas