ros2/rmw_connext #331: Remove reader as argument name

jwillemsen, 2018-12-17:
When removing reader as argument name the `(void) reader;` can be removed

https://github.com/ros2/rmw_connext/blob/99790fa77e76021a62574e2537bc1e30701fd452/rmw_connext_cpp/include/rmw_connext_cpp/connext_static_subscriber_info.hpp#L46

clalancette, 2018-12-17:
I don't really agree; if we remove that `(void) reader`, then we'll get a compiler warning about an unused argument.  Why do you think it can be removed?

jwillemsen, 2018-12-17:
```
virtual void on_subscription_matched(
  DDSDataReader *,
  const DDS_SubscriptionMatchedStatus & status)
{
    current_count_ = status.current_count;
}
```

clalancette, 2018-12-17:
Oh sure.  That is a purely style thing; the generated code won't be any different.  We generally follow the Google C++ [style guide](https://google.github.io/styleguide/cppguide.html), which does agree with your style.  But we have quite a few instances of the C-style `(void) reader`, so I'm not sure this really warrants a change.

jwillemsen, 2018-12-17:
Just something I found when working on rmw_opendds but given the several of pull request already pending I don't want to add more, this one could cause merge conflicts

jwillemsen, 2018-12-17:
Please assign to me

closed 2019-01-09 by jwillemsen