ros2/rmw_connext #334: Shared memory transport still in use

hidmic, 2018-12-17:
## Bug report

**Required Info:**

**Required Info:**
- Operating System:
  - Ubuntu Xenial 16.04 
- Installation type:
  - From sources
- Version or commit hash:
  - Crystal pre-release binaries
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Perform the [QoS demo on lossy networks](https://index.ros.org/doc/ros2/Quality-of-Service/#add-network-traffic) using RTI Connext as your DDS vendor of choice.

#### Expected behavior
As the network degrades, topic subscription rates on a reliable topic degrade as well.
#### Actual behavior
No rate decrease is seen. Moreover, relevant nodes seem to be making use of shared memory.

```sh
------ Shared Memory Creator/Last-op PIDs --------
shmid      owner      cpid       lpid
65536      michel     13438      13495
98305      michel     13438      13495
229378     michel     13495      13438
262147     michel     13495      13495

michel@1bc3650db395:~$ ps ax
 PID TTY      STAT   TIME COMMAND
   1 pts/28   Ss     0:00 /bin/bash
13405 pts/28   S+     0:00 tmux
13407 ?        Ss     0:00 tmux
13408 pts/25   Ss     0:00 -bash
13437 pts/25   S+     0:00 /usr/bin/python3 /opt/ros/crystal/bin/ros2 run image_tools showimage
13438 pts/25   Rl+    0:56 /opt/ros/crystal/lib/image_tools/showimage
13490 pts/29   Ss     0:00 -bash
13494 pts/29   S+     0:00 /usr/bin/python3 /opt/ros/crystal/bin/ros2 run image_tools cam2image
13495 pts/29   Sl+    0:32 /opt/ros/crystal/lib/image_tools/cam2image
13519 pts/30   Ss     0:00 -bash
13532 pts/30   R+     0:00 ps ax
```
#### Additional information
RTI Connext API provides an additional [QoS policy setting](http://community.rti.com/rti-doc/510/ndds/doc/html/api_cpp/structDDS__TransportBuiltinQosPolicy.html) that is not being currently set. Not necessarily related though.

gonzodepedro, 2019-01-07:
I can reproduce this on Ubuntu 18.04 using crystal .deb packages. Also limiting traffic by using tc tool doesn't have any effect in the communication, it seems that shared memory is used when running locally.

gonzodepedro, 2019-01-08:
I was able to reproduce it on Ubuntu 18.04 installing the development environment from master.

gonzodepedro, 2019-01-08:
I added this line:
`  participant_qos.transport_builtin.mask = DDS_TRANSPORTBUILTIN_UDPv4;`
to ros2/rmw_connext/rmw_connext_shared_cpp/src/node.cpp 
And it works as expected. traffic is sent over the loopback interface and the QoS example behaves as expected. Notice that with that line of code no shared memory is ever used.

ivanpauno, 2019-07-15:
@hidmic I think we can close this one, right?

hidmic, 2019-07-15:
Yes!

closed 2019-07-15 by hidmic