ros2/rmw_connext #340: Patching of RTI generated headers/sources very risky

jwillemsen, 2018-12-18:
The rmw_connext_cpp project patches the generated RTI files. This is very risky, this is all RTI internal code and can change between versions. It also makes porting of code between vendors much harder because now vendor specific extensions are created

dirk-thomas, 2018-12-18:
Wer are aware that the current approach is fragile. Do you have an alternative proposal?

jwillemsen, 2018-12-18:
Working slowly on rmw_opendds, hopefully I can make a proposal as part of that

jwillemsen, 2018-12-18:
Is https://github.com/ros2/rosidl_typesupport_connext/pull/15 improving this?

dirk-thomas, 2018-12-18:
> Is ros2/rosidl_typesupport_connext#15 improving this?

No, the IDl pipeline related changes are orthogonal.

clalancette, 2019-01-03:
We don't have an alternative at the moment, so we'll close this out for now.  If you have an alternative to make this better, please feel free to make a pull request.

closed 2019-01-03 by clalancette