ros2/rmw_connext #375: Support for Connext 6

dirk-thomas, 2019-10-16:
Some preliminary work regarding Connext 6 support has landed in #371, ros2/rosidl_typesupport_connext#37, and #374 to allow adding Connext 6 support at any time without breaking anything. The patch file `connext_static_serialized_dataPlugin.cxx.patch` which contains the heavy lifting and still needs to be updated for Connext 6 though.

@Karsten1987 @neil-rti I will leave this up to either of you (or anyone else) to contribute the missing bits in a new pull request. Until then ROS 2 will be limited to only support Connext 5.

With the conditional logic in place such a change can land any time - it just won't change the Connext version we are building against in already released distributions though - since that is specified in [REP 2000](https://www.ros.org/reps/rep-2000.html) and locked before a release.

Karsten1987, 2019-10-16:
I am very much in favor of replacing the custom logic with a more elegant and future prone solution. However, I currently don't have an easy way of doing so and I am currently not able to spend more time on this, I'd kindly ask @neil-rti or somebody with more insights in the Connext 6.0 API to help out.

AndreasAZiegler, 2019-12-10:
May I ask about the current status of Connext 6 support?

ivanpauno, 2019-12-20:
I'm starting with https://github.com/ros2/rmw_connext/issues/381.
A lot of collisions are possible between the tasks, so I recommend waiting for https://github.com/ros2/rmw_connext/issues/381 before starting with this one (less rebasing work).

AFAIK, there is nobody working on this. If that's not the case, let me know ASAP.

AndreasAZiegler, 2020-03-04:
Are there any update?

dirk-thomas, 2020-03-04:
> Are there any update?

No comment implies nothing has changes since the last comments - see above:

* "I'd kindly ask @neil-rti or somebody with more insights in the Connext 6.0 API to help out."
* "AFAIK, there is nobody working on this."

neil-rti, 2020-03-06:
It's a work in process; ideally we'd like to fix the build issues and address the performance bottleneck in the RMW layer for both 5.3 and 6.0 versions.  I can't give a firm ETA at this point.

dirk-thomas, 2020-03-06:
@neil-rti Do you think it is realistic to expect that this will happen and get merged for Foxy (which means before the freeze deadline on April 8th)?

neil-rti, 2020-03-09:
@dirk-thomas Probably not, although we may have contributions to merge for the 5.3.1 RMW layer.

closed 2023-06-21 by clalancette