ros2/rmw_connext #380: Implement the rmw_get_publishers_info_by_topic and rmw_get_subscriptions_info_by_topic methods

jaisontj, 2019-12-11:
As a follow up to [this](https://github.com/ros2/rmw_connext/pull/377) PR which returns `RMW_RET_UNSUPPORTED` for the `rmw_get_publishers_info_by_topic` and `rmw_get_subscriptions_info_by_topic`; ensure that these methods are implemented before the ROS2 F release.

These methods are implemented in `rmw_fastrtps` in [this](https://github.com/ros2/rmw_fastrtps/pull/336) PR which can be used as a reference.

ivanpauno, 2019-12-11:
Assigning myself, same comment as https://github.com/ros2/rmw_opensplice/issues/299#issuecomment-564530042.

wjwwood, 2020-01-21:
When this is done, the tests need to be fixed in `rclpy` to actually test things:

https://github.com/ros2/rclpy/pull/454/files#diff-960e40f792ccd92281ee56dfb65726bcR197

mm318, 2020-02-21:
The pull request for resolving this issue has been opened at https://github.com/ros2/rmw_connext/pull/391.

mm318, 2020-02-21:
> When this is done, the tests need to be fixed in `rclpy` to actually test things:
> 
> https://github.com/ros2/rclpy/pull/454/files#diff-960e40f792ccd92281ee56dfb65726bcR197

This change is in pull request https://github.com/ros2/rclpy/pull/511.

closed 2023-06-21 by clalancette