ros2/rmw_connext #388: Unable to build overlays when connext not installed

mikaelarguedas, 2020-02-05:
# Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - nightly archive
- Version or commit hash:
  - da75447b0068627fdd4b30e6ddbb7328afe4567c
- DDS implementation:
  - Fast-RTPS (Connext **not** installed)

#### Steps to reproduce issue

```
docker pull osrf/ros2:nightly
docker run -it --rm osrf/ros2:nightly
mkdir -p ~/ws/src && cd ~/ws/src
git clone https://github.com/ros2/example_interfaces.git
cd ~/ws
colcon build
```

#### Expected behavior

Message package builds successfully

#### Actual behavior

Build fails with the following error:
```
Starting >>> example_interfaces
--- stderr: example_interfaces                         
CMake Error at /opt/ros/foxy/share/rmw_connext_cpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package):
  Found package configuration file:

    /opt/ros/foxy/share/rmw_connext_shared_cpp/cmake/rmw_connext_shared_cppConfig.cmake

  but it set rmw_connext_shared_cpp_FOUND to FALSE so package
  "rmw_connext_shared_cpp" is considered to be NOT FOUND.
Call Stack (most recent call first):
  /opt/ros/foxy/share/rmw_connext_cpp/cmake/rmw_connext_cppConfig.cmake:38 (include)
  /opt/ros/foxy/share/rmw_implementation_cmake/cmake/call_for_each_rmw_implementation.cmake:61 (find_package)
  /opt/ros/foxy/share/rosidl_generator_py/cmake/rosidl_generator_py_get_typesupports.cmake:26 (call_for_each_rmw_implementation)
  /opt/ros/foxy/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:27 (rosidl_generator_py_get_typesupports)
  /opt/ros/foxy/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /opt/ros/foxy/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:16 (rosidl_generate_interfaces)

---
Failed   <<< example_interfaces	[ Exited with code 1 ]

```

#### Additional information

Faiklure seems to have appeared at the same time #385 was merged

dirk-thomas, 2020-02-06:
Thanks for the report. Please see #389 for the proposed fix. It would be great if you could double check that it works for you too.

closed 2020-02-06 by dirk-thomas

mikaelarguedas, 2020-02-06:
Just tried this, it built successfully :+1:

PR #389 (2020-02-06): finding rmw_connext_shared_cpp must succeed, only rmw_connext_cpp can signal not-found when Connext is not available
Fixes #388 which was introduced by #385.

The CI builds won't show the specific problem or if it was resolved. Therefore only a single sanity check on Linux: [image]

Packaging build to try the archive: https://ci.ros2.org/job/ci_packaging_linux/294/