ros2/rmw_connext #390: rti-connext-dds package requires user interaction during installation

christian-rauch, 2020-02-14:
## Bug report

I am not sure where to report this issue. Feel free to move this issue to another repo.

The package [`rti-connext-dds-5.3.1_5.3.1-nc.x64Linux3gcc5.4.0+2_amd64.deb`](http://packages.ros.org/ros2/ubuntu/pool/main/r/rti-connext-dds-5.3.1/rti-connext-dds-5.3.1_5.3.1-nc.x64Linux3gcc5.4.0+2_amd64.deb) requires user interaction during installation. This breaks most GUI or automated package management.

Please stop providing Debian packages that require user interaction.

**Required Info:**

- Operating System: Ubuntu 18.04
- Installation type: binaries
- Version or commit hash: 5.3.1
- DDS implementation: RTI Connext

#### Steps to reproduce issue
1. `sudo apt install rti-connext-dds-5.3.1`

#### Expected behavior
Installation of packages should finish automatically.

#### Actual behavior
Package waits for user input.

dirk-thomas, 2020-02-14:
The Debian package is provided by RTI. The interactive confirmation of the license is intentional.

You can automate the acceptance by defining the environment variable `RTI_NC_LICENSE_ACCEPTED=yes`.

closed 2020-02-14 by dirk-thomas

ruffsl, 2020-02-14:
FYI, an example headless install in Docker:
https://github.com/osrf/docker_images/blob/90eaa0d9355886ad56acfd52e5c554e653200411/ros2/nightly/nightly-rmw-nonfree/Dockerfile#L5

Kjwon15, 2021-04-02:
It should be notice about that variable and fail the installation process when `DEBIAN_FRONTEND=noninteractive`.
It is apt's standard operation
FYI: https://www.cyberciti.biz/faq/explain-debian_frontend-apt-get-variable-for-ubuntu-debian/

realtime-neil, 2023-04-04:
@tribela is correct.

https://answers.ros.org/question/373343/how-to-stop-rti-connext-dds-531-prompting-during-installation/

By prompting for user interaction even when `DEBIAN_FRONTEND=noninteractive`, this package violates Debian policy.

https://www.debian.org/doc/debian-policy/ch-binary.html#s-maintscriptprompt