ros2/rmw_connext #410: Build error in Windows CI

rotu, 2020-04-17:
For some reason, `rtiddsgen_server.bat` does not seem to be found.

https://ci.ros2.org/job/ci_windows/10157/consoleText

```
--- output: rmw_connext_cpp
Not searching for unused variables given on the command line.
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.18363.
-- The C compiler identification is MSVC 19.25.28614.0
-- The CXX compiler identification is MSVC 19.25.28614.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.8.2 (C:/ci/ws/install/share/ament_cmake/cmake)
-- Found PythonInterp: C:/Python37/python.exe (found suitable version "3.7.6", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: C:/Python37/python.exe
-- Found connext_cmake_module: 0.8.4 (C:/ci/ws/install/share/connext_cmake_module/cmake)
-- Found RTI Connext: C:/Program Files/rti_connext_dds-5.3.1
-- Found rcutils: 0.8.4 (C:/ci/ws/install/share/rcutils/cmake)
-- Found rosidl_typesupport_connext_c: 0.8.4 (C:/ci/ws/install/share/rosidl_typesupport_connext_c/cmake)
-- Found rosidl_adapter: 0.8.2 (C:/ci/ws/install/share/rosidl_adapter/cmake)
-- Found rmw_connext_shared_cpp: 0.8.1 (C:/ci/ws/install/share/rmw_connext_shared_cpp/cmake)
-- Found ament_lint_auto: 0.8.1 (C:/ci/ws/install/share/ament_lint_auto/cmake)
-- Added test 'copyright' to check for copyright in CMake / C / C++ / Python code
-- Added test 'cppcheck' to perform static code analysis on C / C++ code
-- Configured cppcheck include dirs: C:/ci/ws/src/ros2/rmw_connext/rmw_connext_cpp/include
-- Added test 'cpplint' to check C / C++ code against the Google style
-- Added test 'flake8' to check Python code syntax and style conventions
-- Added test 'lint_cmake' to check CMake code style
-- Added test 'pep257' to check Python code against some of the style conventions in PEP 257
-- Added test 'uncrustify' to check C / C++ code style
-- Added test 'xmllint' to check XML markup files
-- Configuring done
-- Generating done
-- Build files have been written to: C:/ci/ws/build/rmw_connext_cpp
Microsoft (R) Build Engine version 16.5.0+d4cbfca49 for .NET Framework

Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System

  Generating serialized type support for RTI Connext (using 'rtiddsgen_server.bat')

  'rtiddsgen_server.bat' is not recognized as an internal or external command,

  operable program or batch file.

C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(231,5): error MSB6006: "cmd.exe" exited with code 1. [C:\ci\ws\build\rmw_connext_cpp\rmw_connext_cpp.vcxproj]

---
Failed   <<< rmw_connext_cpp	[ Exited with code 1 ]
```

emersonknapp, 2020-04-30:
I am also seeing this (repeated on subsequent run, not a one-off) https://ci.ros2.org/job/ci_windows/10472/console when trying to build Eloquent on the buildfarm.

From the CI runs on this PR https://github.com/ros2/rosbag2/pull/391

clalancette, 2020-12-16:
I don't believe we have seen this particular problem in a long time, so I'm going to close this out.  Feel free to reopen if this is still an issue.

closed 2020-12-16 by clalancette