ros2/rmw_connext #418: Trying RTI DDS TCP in ROS 2, but failed

KaeYan, 2020-04-28:
Error report:
`DDS_ResourceLimitsQosPolicy_is_consistent_w_historyI:inconsistent QoS policies: history.depth and resource_limits.max_samples (max_samples_per_instance is unlimited)
DDS_Subscriber_create_datareader_disabledI:ERROR: Inconsistent QoS
DDSDataReader_impl::create_disabledI:!create reader
DDSDataReader_impl::createI:!create reader
initialize:!create DataReader
connext::details::EntityUntypedImpl::initialize:!failed (see previous errors)

[rcutils|error_handling.c:107] rcutils_set_error_state()
This error state is being overwritten:

  'C++ exception during construction of Requester, at /tmp/binarydeb/ros-eloquent-rcl-interfaces-0.8.0/obj-x86_64-linux-gnu/rosidl_typesupport_connext_cpp/rcl_interfaces/srv/dds_connext/get_parameters__type_support.cpp:704'

with this new error message:

  'failed to create replier, at /tmp/binarydeb/ros-eloquent-rmw-connext-cpp-0.8.1/src/rmw_service.cpp:138'

rcutils_reset_error() should be called after error handling to avoid this.

terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  could not create service: failed to create replier, at /tmp/binarydeb/ros-eloquent-rmw-connext-cpp-0.8.1/src/rmw_service.cpp:138, at /tmp/binarydeb/ros-eloquent-rcl-0.8.4/src/rcl/service.c:179`

**Required Info:**

- Operating System:
  Ubuntu 18.04
- Installation type:
  sudo apt install ros-eloquent-rmw-connext-cpp
- Version or commit hash:
  5.3
- DDS implementation:
  RTI Connext
- Client library (if applicable):
  rclcpp

#### Steps to reproduce issue
`publisher_ = this->create_publisher<std_msgs::msg::String>("topic", rclcpp::SystemDefaultsQoS());`

`export NDDS_QOS_PROFILES=/home/kaeyan-1804/test_exe_ws/src/test_exe/USER_QOS_PROFILES.xml`

[test_exe_ws.zip](https://github.com/ros2/rmw_connext/files/4543162/test_exe_ws.zip)

#### Additional information
I need the RTI in ROS 2 to create a tcp client with the QOS file. The QOS file I have tested with regular RTI DDS application, it works. Furthermore, do you have any example code to configure the RTI in ROS 2 to create a TCP client connecting to an RTI ROS 2 TCP server? Thanks.

ivanpauno, 2020-06-12:
> `DDS_ResourceLimitsQosPolicy_is_consistent_w_historyI:inconsistent QoS policies: history.depth and resource_limits.max_samples (max_samples_per_instance is unlimited)
DDS_Subscriber_create_datareader_disabledI:ERROR: Inconsistent QoS
DDSDataReader_impl::create_disabledI:!create reader
DDSDataReader_impl::createI:!create reader
initialize:!create DataReader
connext::details::EntityUntypedImpl::initialize:!failed (see previous errors)

I've been able to reproduce the problem.
Somehow, data reader resource limit max_samples is being set to `256`, instead of the default `-1`.

I have to investigate more, I'm not pretty sure of what's happening.

ivanpauno, 2020-06-12:
@KaeYan ROS2 always uses the default qos profile, so the `Client_Profile` and `Server_Profile` you created aren't going to be used.

ivanpauno, 2020-06-12:
The default `max_samples` for the parent profile you're specifying (`BuiltinQosLibExp::Generic.StrictReliable`) is `256`.
One of the parameter services that are automatically created uses a history depth of 1000.

If you're not interested in changing parameters from an external process, you can disable those services (if not, they are automatically created).
You can also pass a custom qos to them with a smaller history depth.

If not, you can add this to your QoS profile file:

```xml
      <datareader_qos>
        <resource_limits>
          <max_samples>-1</max_samples>
          <max_samples_per_instance>-1</max_samples_per_instance>
        </resource_limits>
      </datareader_qos>
```

closed 2020-06-12 by ivanpauno