ros2/rmw_connext #435: Error getting RMW implementation identifier

fujitatomoya, 2020-07-01:
## Bug report

- Operating System:
  - Ubuntu20.04
- Installation type:
  - Source build
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/0ee1ca223bcc7185743da1ecccc754b774cd0958
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - rclcpp/rclpy

#### Steps to reproduce issue

according to the procedure: https://index.ros.org/doc/ros2/Installation/DDS-Implementations/#linux-binary-install

```
# cd <colcon_ws>
# source install/local_setup.bash

# dpkg -l | grep rti-connext
ii  rti-connext-dds-5.3.1                5.3.1-nc.x64Linux3gcc5.4.0+2         amd64        RTI Connext DDS 5.3.1 tools and libraries

# ls /opt/rti.com/rti_connext_dds-5.3.1/rti_license.dat
/opt/rti.com/rti_connext_dds-5.3.1/rti_license.dat

# export RTI_LICENSE_FILE=/opt/rti.com/rti_connext_dds-5.3.1/rti_license.dat

# cd /opt/rti.com/rti_connext_dds-5.3.1/resource/scripts && source ./rtisetenv_x64Linux3gcc5.4.0.bash; cd -
###############################################################################

         (c) Copyright, Real-Time Innovations, All rights reserved.

                           RTI Connext DDS 5.3.1

###############################################################################
/root/ros2_ws/cft_ws

# RMW_IMPLEMENTATION=rmw_connext_cpp; export RMW_IMPLEMENTATION

# ros2 run demo_nodes_cpp talker
[ERROR] [1593583579.151130085] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_connext_cpp'), with error message 'failed to find shared library of rmw implementation. Searched rmw_connext_cpp, at /root/ros2_ws/cft_ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:54', exiting with 1.
```

#### Expected behavior

```
# ros2 run demo_nodes_cpp talker
[INFO] [1593583821.244770803] [talker]: Publishing: 'Hello World: 1'
<snip>
```

#### Actual behavior

```
# ros2 run demo_nodes_cpp talker
[ERROR] [1593583579.151130085] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_connext_cpp'), with error message 'failed to find shared library of rmw implementation. Searched rmw_connext_cpp, at /root/ros2_ws/cft_ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:54', exiting with 1.
```

iuhilnehc-ynos, 2020-07-01:
@fujitatomoya 

`RMW implementation not installed`
Could you install `ros-foxy-rmw-connext-cpp` first? and try it again.

fujitatomoya, 2020-07-01:
I resolved the problem with the following procedure,

```
# export RTI_LICENSE_FILE=/opt/rti.com/rti_connext_dds-5.3.1/rti_license.dat
# cd /opt/rti.com/rti_connext_dds-5.3.1/resource/scripts && source ./rtisetenv_x64Linux3gcc5.4.0.bash; cd -
# colcon build --symlink-install
...<snip>
# RMW_IMPLEMENTATION=rmw_connext_cpp; export RMW_IMPLEMENTATION
# source install/local_setup.bash
# ros2 run demo_nodes_cpp talker
RTI Data Distribution Service Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00 See www.rti.com for more information.
[INFO] [1593590317.639785150] [talker]: Publishing: 'Hello World: 1'

```

RTI setting is required before starting build the source code, that will build rmw_connext_*.

closing this issue.

closed 2020-07-01 by fujitatomoya