ros2/rmw_connext #476: [Requesting Help] Support ContentFilteredTopic on rmw_connext

fujitatomoya, 2020-11-12:
This is NOT exactly issue or bug in current source code, but we are not sure where should be the right place to discuss details about implementation. If this is inappropriate place, could you kindly guide us.

## Description

### Background

We Sony are now working on to enable [ContentFilteredTopic](https://community.rti.com/static/documentation/connext-dds/5.3.0/doc/api/connext_dds/api_cpp2/classdds_1_1topic_1_1ContentFilteredTopic.html) on rmw_connext based on [ros2/design for ContentFilteredTopic](https://github.com/ros2/rmw/issues/241). Being optional to have this feature relies on the RMW implementation can lead a huge efficiency for network and CPU resource especially on edge IoT devices. Currently RTI Connext is the one to support [ContentFilteredTopic](https://community.rti.com/static/documentation/connext-dds/5.3.0/doc/api/connext_dds/api_cpp2/classdds_1_1topic_1_1ContentFilteredTopic.html), so we'd like to enable this feature based on RTI Connext.

### Problem

Since there is specific dds writer/reader exists in,

https://github.com/ros2/rmw_connext/blob/37117a9b332eee697bc058d8f5fbdac299e464c3/rmw_connext_cpp/src/rmw_publish.cpp#L31-L32

and

https://github.com/ros2/rmw_connext/blob/37117a9b332eee697bc058d8f5fbdac299e464c3/rmw_connext_cpp/src/rmw_take.cpp#L75-L76

we are unable to use ContentFilteredTopic so far. If we use ContentFilteredTopic with filtering expression and expression parameters in reader side, `rmw_take` does not get event at all. It seems that [ConnextStaticSerializedData](https://github.com/ros2/rmw_connext/blob/master/rmw_connext_cpp/resources/connext_static_serialized_data.idl) has been introduced via https://github.com/ros2/rmw_connext/pull/259. But we don't quiet understand how it works and why it is needed. Could anyone please elaborate a little bit?

### Note

- We've confirmed that we can use [ContentFilteredTopic](https://community.rti.com/static/documentation/connext-dds/5.3.0/doc/api/connext_dds/api_cpp2/classdds_1_1topic_1_1ContentFilteredTopic.html) on RTI Connext w/o any problems.

fujitatomoya, 2020-11-12:
CC: @Barry-Xu-2018 @iuhilnehc-ynos

fujitatomoya, 2020-11-12:
@Karsten1987 
CC: @ivanpauno @wjwwood 

It seems that you are the original author of https://github.com/ros2/rmw_connext/pull/259, would you mind explaining a little kindly? This's been blocking us for days, we'd really appreciate for your help!

Barry-Xu-2018, 2020-11-12:
@Karsten1987

Now we are implementing the feature of content filtered Topic. DDS is RTI connext for verification. 

We encountered an issue that `topic_reader` created by a subscriber with `content_filtered_topic` can't not receive messages from a publisher of ROS2. `read_condition` is not be triggered by `DDS::WaitSet`.  

After investigation, we find it seems to be related to #259. (During investigation, we created a issue #474. We think #474 is caused by #259).     
We don't know the detail of implementation of RTi connext.   
Below is our thought.  We want to discuss with you to find out the solution to support content filtered topic.  

```
1. rti sample (standard method)

# writer to filter a topic (use a data type, for example string.idl with a key member)

connext

    reader side
        set filter: key = 1

    writer side
        write_msg("key=1")
            # internal implementation (It's just guess.)
            for loop to check if the peer reader is filtered by key== 1
            if yes
                write_to_a_specific_reader
            else
                do nothing

2. ros2 (current)

connext

    reader side
        read_condition created by normal topic reader

    writer side
        write_msg(serialization_data(raw_data about "key=1"))
            just use the narrowed writer's socket to send data. (so the read_condition can be triggered)

3. ros2 with content filter topic feature (new feature)

# writer to filter a topic (data type: connext_static_serialized_data.idl)

connext

    reader side
        set filter: key = 1
        read_condition created by cft topic reader. 
        (We guess read_condition will be triggered inside only after receiving the filtered data, 
         it's totally different from the read_condition created by normal topic.)

    writer side
        write_msg(serialization_data(raw_data about "key=1"))
            Here is an issue.
            We think that it uses current writer's socket to send data
            and it will not check which peer reader this writer send because 
            this writer knows nothing about the raw_data "key=1"
```

fujitatomoya, 2020-11-16:
@Karsten1987 

friendly ping, we got really stuck here. in the meantime, we will be searching the solution.

fujitatomoya, 2021-01-27:
we will give it a shot with https://github.com/rticommunity/rmw_connextdds/ instead.

fujitatomoya, 2021-03-11:
closing this, since https://github.com/rticommunity/rmw_connextdds/ works okay.

closed 2021-03-11 by fujitatomoya